eez-open/modular-psu-firmware #6: New feature: AC-mode

AndersWestman, 2019-07-12:
Would it be possible to make an “AC-mode” there a predefined list or similar is doing the half wave for each channel. The channels are external serial coupled to make a midpoint reference.  
If it’s possible to calculate the list in the firmware just by typing in AC voltage and frequency that would be great, but some predefined voltages at 50 and 60Hz should possible cover many cases.

prasimix, 2019-07-24:
We have to add first the new set of SCPI commands as a building blocks for this functionality that in combination with new `SRAil` (see [INSTrument:COUPle:TRACking](https://www.envox.hr/eez/eez-bench-box-3/bb3-scpi-reference-manual/5-subsystem-command-reference/bb3-scpi-instrument.html#inst_coup)) type of output coupling could do the job: 

- `:AMPlitude {<amplitude>}` Sets the arbitrary waveform amplitude
- `:COUNt {<count>}` Sets the number of arbitrary waveforms
- `:FREQuency {<frequency>}` Sets the arbitrary waveform frequency
- `:FUNCtion {<type>}` Sets the arbitrary waveform parameter type
- `:OFFset {<level>}` Sets the offset level of the arbitrary waveform
- `:PHASe[:ADJust] {<angle>}` Set the arbitrary waveform start phase
- `:SHAPe {<waveform>}` Selects the shape of the arbitrary waveform

prasimix, 2019-11-21:
Cannot be done without 2-quadrant outputs. The DCP405 which has down-programmer has limited current sink capability.

closed 2019-11-21 by prasimix