eez-open/modular-psu-firmware #15: Build issues with Atollic TrueStudio for STM32 v9.3.0 on Windows7

bvernoux, 2020-01-06:
I'm using windows 7 Pro 64bits SP1 + Atollic TrueStudio for STM32 v9.3.0 (Last version available)

Step to reproduce the issue:
- I have cloned this repository https://github.com/eez-open/modular-psu-firmware.git to my directory D:\_proj\__Lab_Tools\PowerSupply\EEZ-BB3\
- Atollic TrueStudio for STM32 v9.3.0 steps:
  - I select File->Import... -> Existing Projects into Workspace ->Select root directory: D:\_proj\__Lab_Tools\PowerSupply\EEZ-BB3\modular-psu-firmware\src\third_party\stm32_truestudio
  - I click on Finish (bb3 project is found)
  - I right click on bb3 project -> Build project

I obtain the error:
```
07:58:38 **** Rebuild of configuration Debug for project bb3 ****
Info: Internal Builder is used for build
arm-atollic-eabi-gcc -c -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -std=gnu11 -DUSE_HAL_DRIVER -DEEZ_BP3C_REVISION_R3B1=1 -DOPTION_SDRAM=1 -DLWIP_NETIF_HOSTNAME=1 -DSTM32F769xx -DEEZ_PLATFORM_STM32 -DDEBUG -DLZ4_FORCE_MEMORY_ACCESS=0 -DSCPI_USER_CONFIG=1 -DOPTION_ETHERNET=1 -DOPTION_EXT_RTC=1 -DOPTION_SD_CARD=1 -DOPTION_BP=0 -DOPTION_DISPLAY=1 -DOPTION_FAN=1 -DOPTION_AUX_TEMP_SENSOR=1 -DOPTION_WATCHDOG=1 -DOPTION_ENCODER=1 -DOPTION_EXT_EEPROM=1 -DEEZ_MCU_REVISION_R1B5=1 -DLODEPNG_NO_COMPILE_ALLOCATORS=1 -DLODEPNG_NO_COMPILE_DECODER=1 -DLODEPNG_NO_COMPILE_DISK=1 -DLODEPNG_NO_COMPILE_ANCILLARY_CHUNKS=1 -DLODEPNG_NO_COMPILE_ERROR_TEXT=1 -DLODEPNG_NO_COMPILE_CPP=1 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -I../Inc -I../Middlewares/Third_Party/LwIP/src/include -I../Middlewares/Third_Party/LwIP/system -I../Drivers/STM32F7xx_HAL_Driver/Inc -I../Drivers/STM32F7xx_HAL_Driver/Inc/Legacy -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1 -I../Middlewares/Third_Party/LwIP/src/include/netif/ppp -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F7xx/Include -I../Middlewares/Third_Party/LwIP/src/include/lwip -I../Middlewares/Third_Party/LwIP/src/include/lwip/apps -I../Middlewares/Third_Party/LwIP/src/include/lwip/priv -I../Middlewares/Third_Party/LwIP/src/include/lwip/prot -I../Middlewares/Third_Party/LwIP/src/include/netif -I../Middlewares/Third_Party/LwIP/src/include/posix -I../Middlewares/Third_Party/LwIP/src/include/posix/sys -I../Middlewares/Third_Party/LwIP/system/arch -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../.. -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../libscpi/inc -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../micropython -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../micropython/ports/bb3 -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/.. -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../../eez/scpi -ID:\_proj\__Lab_Tools\PowerSupply\EEZ-BB3\modular-psu-firmware\src\third_party\stm32_truestudio\Middlewares\Third_Party\FatFs\src -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FatFs/src -Og -ffunction-sections -fdata-sections -g -fstack-usage -Wall -o Middlewares\Third_Party\FreeRTOS\Source\event_groups.o ..\Middlewares\Third_Party\FreeRTOS\Source\event_groups.c 
arm-atollic-eabi-gcc -c -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -std=gnu11 -DUSE_HAL_DRIVER -DEEZ_BP3C_REVISION_R3B1=1 -DOPTION_SDRAM=1 -DLWIP_NETIF_HOSTNAME=1 -DSTM32F769xx -DEEZ_PLATFORM_STM32 -DDEBUG -DLZ4_FORCE_MEMORY_ACCESS=0 -DSCPI_USER_CONFIG=1 -DOPTION_ETHERNET=1 -DOPTION_EXT_RTC=1 -DOPTION_SD_CARD=1 -DOPTION_BP=0 -DOPTION_DISPLAY=1 -DOPTION_FAN=1 -DOPTION_AUX_TEMP_SENSOR=1 -DOPTION_WATCHDOG=1 -DOPTION_ENCODER=1 -DOPTION_EXT_EEPROM=1 -DEEZ_MCU_REVISION_R1B5=1 -DLODEPNG_NO_COMPILE_ALLOCATORS=1 -DLODEPNG_NO_COMPILE_DECODER=1 -DLODEPNG_NO_COMPILE_DISK=1 -DLODEPNG_NO_COMPILE_ANCILLARY_CHUNKS=1 -DLODEPNG_NO_COMPILE_ERROR_TEXT=1 -DLODEPNG_NO_COMPILE_CPP=1 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -I../Inc -I../Middlewares/Third_Party/LwIP/src/include -I../Middlewares/Third_Party/LwIP/system -I../Drivers/STM32F7xx_HAL_Driver/Inc -I../Drivers/STM32F7xx_HAL_Driver/Inc/Legacy -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1 -I../Middlewares/Third_Party/LwIP/src/include/netif/ppp -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F7xx/Include -I../Middlewares/Third_Party/LwIP/src/include/lwip -I../Middlewares/Third_Party/LwIP/src/include/lwip/apps -I../Middlewares/Third_Party/LwIP/src/include/lwip/priv -I../Middlewares/Third_Party/LwIP/src/include/lwip/prot -I../Middlewares/Third_Party/LwIP/src/include/netif -I../Middlewares/Third_Party/LwIP/src/include/posix -I../Middlewares/Third_Party/LwIP/src/include/posix/sys -I../Middlewares/Third_Party/LwIP/system/arch -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../.. -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../libscpi/inc -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../micropython -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../micropython/ports/bb3 -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/.. -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../../eez/scpi -ID:\_proj\__Lab_Tools\PowerSupply\EEZ-BB3\modular-psu-firmware\src\third_party\stm32_truestudio\Middlewares\Third_Party\FatFs\src -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FatFs/src -Og -ffunction-sections -fdata-sections -g -fstack-usage -Wall -o micropython\py\showbc.o ..\..\micropython\py\showbc.c 
arm-atollic-eabi-gcc -c -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -std=gnu11 -DUSE_HAL_DRIVER -DEEZ_BP3C_REVISION_R3B1=1 -DOPTION_SDRAM=1 -DLWIP_NETIF_HOSTNAME=1 -DSTM32F769xx -DEEZ_PLATFORM_STM32 -DDEBUG -DLZ4_FORCE_MEMORY_ACCESS=0 -DSCPI_USER_CONFIG=1 -DOPTION_ETHERNET=1 -DOPTION_EXT_RTC=1 -DOPTION_SD_CARD=1 -DOPTION_BP=0 -DOPTION_DISPLAY=1 -DOPTION_FAN=1 -DOPTION_AUX_TEMP_SENSOR=1 -DOPTION_WATCHDOG=1 -DOPTION_ENCODER=1 -DOPTION_EXT_EEPROM=1 -DEEZ_MCU_REVISION_R1B5=1 -DLODEPNG_NO_COMPILE_ALLOCATORS=1 -DLODEPNG_NO_COMPILE_DECODER=1 -DLODEPNG_NO_COMPILE_DISK=1 -DLODEPNG_NO_COMPILE_ANCILLARY_CHUNKS=1 -DLODEPNG_NO_COMPILE_ERROR_TEXT=1 -DLODEPNG_NO_COMPILE_CPP=1 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -I../Inc -I../Middlewares/Third_Party/LwIP/src/include -I../Middlewares/Third_Party/LwIP/system -I../Drivers/STM32F7xx_HAL_Driver/Inc -I../Drivers/STM32F7xx_HAL_Driver/Inc/Legacy -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1 -I../Middlewares/Third_Party/LwIP/src/include/netif/ppp -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F7xx/Include -I../Middlewares/Third_Party/LwIP/src/include/lwip -I../Middlewares/Third_Party/LwIP/src/include/lwip/apps -I../Middlewares/Third_Party/LwIP/src/include/lwip/priv -I../Middlewares/Third_Party/LwIP/src/include/lwip/prot -I../Middlewares/Third_Party/LwIP/src/include/netif -I../Middlewares/Third_Party/LwIP/src/include/posix -I../Middlewares/Third_Party/LwIP/src/include/posix/sys -I../Middlewares/Third_Party/LwIP/system/arch -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../.. -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../libscpi/inc -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../micropython -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular-psu-firmware/src/third_party/stm32_truestudio/../micropython/ports/bb3 -ID:/_proj/__Lab_Tools/PowerSupply/EEZ-BB3/modular [... cut ...] re/src/third_party/stm32_truestudio/../../eez/scpi -ID:\_proj\__Lab_Tools\PowerSupply\EEZ-BB3\modular-psu-firmware\src\third_party\stm32_truestudio\Middlewares\Third_Party\FatFs\src -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FatFs/src -Og -ffunction-sections -fdata-sections -g -fstack-usage -Wall -o Drivers\STM32F7xx_HAL_Driver\Src\stm32f7xx_hal_sdram.o ..\Drivers\STM32F7xx_HAL_Driver\Src\stm32f7xx_hal_sdram.c 
Info: Parallel threads used: 6

07:58:39 Build Finished (took 630ms)
```

prasimix, 2020-05-18:
@bvernoux please try again with the latest version and let us know if the problem still exists or just close this issue.

bvernoux, 2020-05-18:
I confirm with latest commit 6a34e27 and using Atollic TrueStudio for STM32 v9.3.0 (latest available) all is fine now
Issue closed

closed 2020-05-18 by bvernoux