eez-open/modular-psu-firmware #19: SCPI via UART

prasimix, 2020-02-15:
Accepting SCPI commands via UART will enables simple device without USB or Ethernet support to communicate and control BB3

mvladic, 2021-02-19:
Add SCPI command to transmit ASCII string through UART, for example:

`SYSTem:COMMunicate:UART:TRANsmit {string}`

prasimix, 2021-06-06:
Add an example of how to control one BB3 with another via UART (something simple, e.g. setting the voltage and output to ON state).

prasimix, 2021-06-09:
> Add an example of how to control one BB3 with another via UART (something simple, e.g. setting the voltage and output to ON state).

An [example](https://github.com/eez-open/modular-psu-firmware/blob/master/docs/examples/arduino_dlog_bookmarks/arduino_dlog_bookmarks.ino), see also short [video](https://www.youtube.com/watch?v=337BuP4VSWc).

Fixed

prasimix, 2021-06-13:
We need configurable communication params via GUI and SCPI. For SCPI we can use the following commands:

`SYSTem:COMMunicate:UART:BAUD (9600, 19200, 38400, ... )`
`SYSTem:COMMunicate:UART:PARIty`
`SYSTem:COMMunicate:UART:DATA`
`SYSTem:COMMunicate:UART:STOP`

prasimix, 2021-06-15:
Fixed

closed 2021-06-15 by prasimix