eez-open/modular-psu-firmware #41: Add support for DCM224 module

prasimix, 2020-08-06:
The DCM224 is modified version of DCM220 module which include the following:

- [ ] Output voltage up to 24 V
- [ ] Output current up to 4.9 A
- [ ] MCU controlled LT3763 controller switching frequency. We have to implement the following:
    - [ ] Base frequency settings (in kHz), range: 350-600 kHz, default: 500 kHz
    - [ ] Frequency dithering (on/off), range +/-10% of set base frequency. Sweep speed need to be determined.
    - [ ] Counterphase signals for Ch1 and Ch2 LT3763 controllers
- [ ] Power output PWM control (dimming)

chengyabcd, 2020-08-12:
After flashing this program on the DCM220 hardware, does it become a DCM224 module?

prasimix, 2020-08-12:
No, DCM224 is a new module (modified DCM220).

chengyabcd, 2020-08-12:
Excuse me, has the hardware of the DCM224 module been released? Is there a link? Thank you!

prasimix, 2020-08-12:
> Excuse me, has the hardware of the DCM224 module been released? Is there a link? Thank you!

Not yet, but it should be in the next few weeks.

chengyabcd, 2020-08-12:
Okay, thank you very much for your answers!

prasimix, 2020-08-27:
Fixed

closed 2020-08-27 by prasimix