eez-open/modular-psu-firmware #42: Add support for MIO168 module

prasimix, 2020-08-06:


prasimix, 2021-06-25:
Fixed

closed 2021-06-25 by prasimix

mvladic, 2021-06-30:
MIO168 SCPI:

### DIN

`INST (@101)`

```
SOURce:DIGital:RANGe {<pin>}, {<range>}
SOURce:DIGital:RANGe? {<pin>}

    <range>: LOW|HIGH
    <pin>: 1-8

SOURce:DIGital:SPEed {<pin>}, {<speed>}
SOURce:DIGital:SPEed? {<pin>}

    <speed>: FAST|SLOW
    <pin>: 1-2

MEASure:DIGital[:BYTE]? [<channel>]
```

### DOUT

`INST (@102)`

```
SOURce:DIGital:DATA[:BYTE] {<value>}
SOURce:DIGital:DATA[:BYTE]?
```

### AIN

- AIN1: `INST (@103)`
- AIN2: `INST (@104)`
- AIN3: `INST (@105)`
- AIN4: `INST (@106)`

```
SENSe:FUNCtion[:ON] {<function>}[, <channel>]
SENSe:FUNCtion[:ON]? [<channel>]

    <function>: CURRent|VOLTage

SENSe:CURRent[:DC]:RANGe {<range>}[, <channel>]
SENSe:CURRent[:DC]:RANGe? [<channel>]

    <range> for AIN1 (@103) and AIN2 (@104)
	    1: +/- 48 mA

    <range> for AIN3 (@105) and AIN4 (@106)
	    1: +/- 24 mA
	    2: +/- 1.2 A
	    3: +/- 10 A

SENSe:VOLTage[:DC]:RANGe {<range>}[, <channel>]
SENSe:VOLTage[:DC]:RANGe? [channel]

    <range> for AIN1 (@103) and AIN2 (@104)
	    1: +/- 2.4 V
	    2: +/- 24 V
	    3: +/- 240 V

    <range> for AIN3 (@105) and AIN4 (@106)
	    1: +/- 2.4 V
	    2: +/- 12 V

SENSe:NPLCycles
SENSe:NPLCycles?
SENSe:APERture?

    <cycles> range: 0 - 25

SYSTem:LFRequency <frequency>
SYSTem:LFRequency?

    <frequency>: 50 or 60

MEASure[:SCALar]:CURRent[:DC]? [<channel>]
MEASure[:SCALar]:VOLTage[:DC]? [<channel>]
```

### AOUT DAC7760

- AOUT1: `INST (@107)`
- AOUT2: `INST (@108)`

```

SOURce:FUNCtion[:ON] {<function>}
SOURce:FUNCtion[:ON]?

    <function>: VOLTage|CURRent

SOURce:CURRent:RANGe {<range>}
SOURce:CURRent:RANGe?

    <range>:
	    1: 4 mA to 20 mA
	    2: 0 mA to 20 mA
	    3: 0 mA to 24 mA

SOURce:VOLTage:RANGe {<range>}
SOURce:VOLTage:RANGe?

    {<range>}:
	    1: 0 V to +5 V
	    2: 0 V to +10 V
	    3: +/- 5 V
	    4: +/- 10 V

OUTPut[:STATe] {bool}[, <chanlist>]
OUTPut[:STATe]? [<chanlist>]

[SOURce[<n>]]:VOLTage[:LEVel][:IMMediate][:AMPLitude]
[SOURce[<n>]]:VOLTage[:LEVel][:IMMediate][:AMPLitude]?

[SOURce[<n>]]:CURRent[:LEVel][:IMMediate][:AMPLitude]
[SOURce[<n>]]:CURRent[:LEVel][:IMMediate][:AMPLitude]?
```

### AOUT DAC7563

- AOUT3: `INST (@109)`
- AOUT4: `INST (@110)`

```
[SOURce[<n>]]:VOLTage[:LEVel][:IMMediate][:AMPLitude]
[SOURce[<n>]]:VOLTage[:LEVel][:IMMediate][:AMPLitude]?
```

### PWM

- PWM1: `INST (@111)`
- PWM2: `INST (@112)`

```
TODO ...
```