eez-open/modular-psu-firmware #75: Module labeling

prasimix, 2020-10-18:
Reported by @danielbriggs

_Is it possible to label the channels with text? I couldn't find a way, nor any mention in the manual. `DCM220 #4` is a great default... but would be nice to have the option to put useful labels on things that persist through power cycles, (or even profile based?) especially when lots of rails going on._

mvladic, 2020-11-25:
Proposed new SCPI commands and queries for slot, channel and pin labeling:

- `SYSTem:SLOT:LABel {slot}, {label}`
	`{label}` is quoted string, min length is 0, max length is 8. If length is 0 (empty string `""`) then label is reset to the default value.

- `SYSTem:SLOT:LABel? {slot}`

- `SYSTem:SLOT:COLor {slot}, {color}`
	`{color}` is value between 0 and 24. If value is 0 then color is reset to the default value.

- `SYSTem:SLOT:COLor? {slot}`

- `SYSTem:CHANnel:LABel {channel}, {label}`
- `SYSTem:CHANnel:LABel? {channel}`
	`{label}` max length for power channels is 8 and for all other channels is 5.

- `SYSTem:CHANnel:COLor {channel}, {color}`
- `SYSTem:CHANnel:COLor? {channel}`

- `SYSTem:CHANnel:PIN:LABel {channel}, {pin}, {label}`
	This is used for digital inputs and digital outputs on MIO168 where each pin can have different label. Allowed `{pin}` range is from 1 to 8. `{label}` max length is 5.
- `SYSTem:CHANnel:PIN:LABel? {channel}, {pin}`

Color table:

|No.|Hex|Color|
|---|---|---|
|1|#d4c36b|[image]|
|2|#9fdd9a|[image]|
|3|#73bdd8|[image]|
|4|#d971ca|[image]|
|5|#dbafea|[image]|
|6|#adaaf9|[image]|
|7|#f44336|[image]|
|8|#e91e63|[image]|
|9|#9c27b0|[image]|
|10|#673ab7|[image]|
|11|#3f51b5|[image]|
|12|#2196f3|[image]|
|13|#03a9f4|[image]|
|14|#00bcd4|[image]|
|15|#009688|[image]|
|16|#4caf50|[image]|
|17|#8bc34a|[image]|
|18|#cddc39|[image]|
|19|#ffeb3b|[image]|
|20|#ffc107|[image]|
|21|#ff9800|[image]|
|22|#ff5722|[image]|
|23|#795548|[image]|
|24|#607d8b|[image]|

prasimix, 2021-01-30:
Move this option from `System info` page to `Display` page

prasimix, 2021-02-01:
When choosing a color, we should mark which color is currently selected.

prasimix, 2021-02-01:
Fixed

fietser28, 2021-02-11:
If I have custom label names set (DCP405 in slot 1 and 2): After turning series coupling on and off the custom label names are forgotten/reset to the default module names.

Tested on 1.6-RC.

prasimix, 2021-02-12:
Fixed

closed 2021-02-12 by prasimix