eez-open/modular-psu-firmware #78: Bargraphs scaling

mvladic, 2020-10-20:
Reported by @MaxZ

> speaking of bargraphs - yet another idea (sorry for spamming you so much guys :sweat_smile:) i like them, just like the plots - seriously. Hooowever... I don't like that its scaling is fixed to the "limit" value. since that is a feature i usually dont use, I'd like to disable it, which means i put the limit to 40V/5A - but if I want to plot something like LED current at 2V, 20mA - i can't see anything on the graph. Wouldn't it be possible to make it auto scaling? or even better, give us a scaling menu.

prasimix, 2020-10-27:
In channel's `Display view` add new selection: `Limit` (what we currently have), `Auto` (a new default) and `Custom` (when user can enters range). In `Auto` mode for better visibility add e.g. 5-10% over set value, for example for U set to 10 V, set range to 10.5 or 11 V.

prasimix, 2021-01-29:
Fixed, `Display view settings` page new look:

[image]

[image]

prasimix, 2021-02-02:
SCPI commands for Full scale is missing. Proposed command is `INSTrument:DISPlay:SCALe {<n>}, {<scale_range>}` where `n` is 1 or 2, and `scale_range` could be `MAXimum`, `LIMit`, `AUTO`, `CUSTom`.

prasimix, 2021-02-02:
Fixed

closed 2021-02-02 by prasimix

fietser28, 2021-02-11:
This change is not reflected in the user manual paragraph 10.1 (https://www.envox.hr/eez/eez-bench-box-3/bb3-user-manual/10-eez-bb3-channel-protections.html):

"Changing the limit will also affect graphical views (vertical bar, horizontal bar and YT views), which will increase readability. For example, if the limit is set to 1 A, this value will become 100% of the scale instead of only 20% when the limit is at 5 A."

Should probably now be:

"Changing the limit will also affect graphical views (vertical bar, horizontal bar and YT views), if the display settings of the channel is set to Limit. For example, if the limit is set to 1 A, this value will become 100% of the scale instead of only 20% when the limit is at 5 A."

Don't know about the auto setting.

prasimix, 2021-02-12:
> 
> "Changing the limit will also affect graphical views (vertical bar, horizontal bar and YT views), if the display settings of the channel is set to Limit. For example, if the limit is set to 1 A, this value will become 100% of the scale instead of only 20% when the limit is at 5 A."
> 
Fixed