eez-open/modular-psu-firmware #85: PNG for Screenshots

MMMZZZZ, 2020-10-21:
I guess the JPEG has been chosen because its natively by the microcontroller. However, it's really not the best format for screenshots. Sharp edges, thin lines - things that are frequent in screenshots - result in tons of artifacts in the compressed image. PNG on the other hand has a lossless compression. Bigger file but absolutely no artifacts and hence the preferred format for screenshots. 

If it's possible, it would be nice to change to PNG somewhere in the future.

mvladic, 2020-10-21:
We didn't find satisfactory PNG encode/decode library. It needs to be small in code size, fast in execution and small in memory consumption, preferably without dynamic memory allocation.

fietser28, 2021-06-08:
https://lodev.org/lodepng/
is this an option?

mvladic, 2021-06-10:
This library is using dynamic memory allocations. I made a decode and encode test program. Here is the source code, image used for decode, resulting image from encode and text file with the information of memory allocations used for decode and encode:

[lodepng_test.zip](https://github.com/eez-open/modular-psu-firmware/files/6629218/lodepng_test.zip)

For the decode of 480x272 image the library is using max. 783632 bytes of memory, and for the encode of the same image it is using max. 800896 bytes - interestingly, it leaks 44405 bytes in the encode.

So, we could use this library if we are using SDRAM memory, but we must develop custom memory allocations library or find some existing library. I have no experience in this, so any help is welcome.