eez-open/modular-psu-firmware #92: Insecure functions should not be used

bvernoux, 2020-10-23:
In actual code there's tons of call to Insecure/unsafe function like sprintf() which seems to do buffer overflow (detected when you build with latest  STM32CubeIDE 1.4.2)

> ..\..\..\eez\action_impl.cpp:1176:6: warning: '%s' directive writing up to 255 bytes into a region of size 244 [-Wformat-overflow=]

I cannot build after as there is an issue in the stm32cube project see issue https://github.com/eez-open/modular-psu-firmware/issues/91

1) Fix issues with open source tools
   * Fix all warnings found by GCC ARM with STM32CubeIDE 1.4.x (as I'm not sure you are using it and you are probably using an older GCC ARM which does not detect anything)
   * I heavily recommend to check the whole code with [CppCheck ](http://cppcheck.sourceforge.net)(latest version)
     * It could be added to the project with STM32CubeIDE (Eclipse CppCheck plugins)

2) Apply good coding practices like (for mid/long term)
   * https://rules.sonarsource.com/c/type/Security Hotspot
   * https://rules.sonarsource.com/cpp/type/Security Hotspot
   * https://rules.sonarsource.com/c/type/Vulnerability
   * https://rules.sonarsource.com/cpp/type/Vulnerability
   * https://rules.sonarsource.com/c/type/Bug
   * https://rules.sonarsource.com/cpp/type/Bug
   * https://rules.sonarsource.com/c/type/Code Smell
   * https://rules.sonarsource.com/cpp/type/Code Smell

mvladic, 2020-10-24:
https://www.approxion.com/dangerously-confusing-interfaces-iv-the-perils-of-cs-safe-string-functions/

mvladic, 2021-01-31:
- Unsecure functions sprintf, strcpy and strcat replaced with secure counterparts (except in the third party code).
- STM32CubeIDE project, Linux and Windows simulators now compiles without warnings (except in the third party code).

bvernoux, 2021-01-31:
Very nice work thanks for the refactor

prasimix, 2021-01-31:
Fixed for now.

closed 2021-01-31 by prasimix