eez-open/modular-psu-firmware #97: Current readout in YT-view and dark theme could be more readable

Cat-Ion, 2020-10-25:
When viewed from an angle, the current readout in YT-view can be too hard to read, while everything is fine:

[image]

This is also the case with lightness and brightness turned to their maximum values. I realize that the light is in a... somewhat pessimal position for this, but since everything else is readable I figured it'd be worth a mention. Just making it the same colour as 'OFF' would probably be fine.

prasimix, 2020-10-26:
Fixed

closed 2020-10-26 by prasimix

kliment, 2020-10-26:
Hmm, how does that commit fix this issue? It seems it was meant to fix another one.

mvladic, 2020-10-26:
For the reference, I was mistakenly commited https://github.com/eez-open/modular-psu-firmware/commit/44c87cadac2e0811c89da14991cec917d4651f7e with description "#97" 6 days ago which is unrelated to this issue.