eez-open/modular-psu-firmware #106: lack of precision in low current range

gojimmypi, 2020-10-31:
As [requested on discord](https://discord.com/channels/511131497801973761/511188695005659157/772164089984057394), there seems to be a problem with current display values for low-range values.

Changing output voltage appears to increment displayed current only in 0.5mA increments on the DCP405, but as [mentioned on discord](https://discord.com/channels/511131497801973761/511188695005659157/772167090823757846):

> Forget DCM for curve tracer, but DCP yes, we made that example for it, and it should easily measure current in uA, not just mA

Here's an example screen snip:

[image]

Cat-Ion, 2020-11-01:
Probably introduced by [this diff](https://github.com/eez-open/modular-psu-firmware/commit/a945439e11d6a3f939459b5b93559db2974b3f54#diff-c0e4eadd23fcb06eb06f657faaf5bfd5c4c236f3ffb490061f2bf0aed52348b3L836)

prasimix, 2020-11-02:
Fixed

closed 2020-11-02 by prasimix