eez-open/modular-psu-firmware #116: SCPI DLOG settings change back to default after each trigger.

fietser28, 2020-11-07:
The settings for functions to log, DLOG sample interval, DLOG time to log and DLOG trigger source are forgotten/set to default after each trigger. I don't think this is correct behaviour.

Example 1:
- First read the values after startup (default values),
- Set the values to something not equal to default.
- Trigger a data logging
- Read the values again: all are reset to default.

[image]

Example 2:
[image]

prasimix, 2020-11-11:
Currently we intentionally reset dlog settings after dlog finished. We'll change that, and use `*RST` to perform reset of dlog settings.

mvladic, 2021-01-29:
Also, we are adding SCPI command `SENSe:DLOG:CLEar` to reset only DLOG settings.

All existing MicroPython scripts that are doing DLOG will be modified to call this SCPI command.

fietser28, 2021-01-29:
Scripts were changed, but there version numbers not increased. Was this intentional?

mvladic, 2021-01-29:
The version number is increased in the scripts-catalog.json. Where else was it supposed to increase?

prasimix, 2021-02-02:
Fixed

closed 2021-02-02 by prasimix