eez-open/modular-psu-firmware #117: DLOG configuration menu & corresponding SCPI settings do not match

fietser28, 2020-11-08:
This items relates to, but is different from #116 

1. If I set period, duration & functions to log using SCPI commands the settings in the menu won't update to the values set by SCPI. 
2. The other way around also doesn't match:  Values set in the menu don't syncronise with the SCPI ? values.

Example for 1 (set with SCPI, read on display):
[image]

Example  for 2 (set on touchscreen, read with SCPI):
[image]

prasimix, 2020-11-11:
Dlog GUI is subject to major redesign due to new modules such as MIO168. The new GUI will take into account suggestions mentioned here, i.e. that when dlog form is opened we have to display previously set values via SCPI or GUI. Also when dlog form is opened eventual new changes via SCPI will be ignored.

prasimix, 2021-01-30:
Sending this sequence results in touchscreen freezing, but firmware is still alive, since communication via USB with Studio is still possible:

[image]

fietser28, 2021-01-31:
As discussed on Discord: Please add an OK button which will save changes and closes the page and will not start log.

The current page only allows for CANCEL (changed are discarded) or START (save changes and start log). 

[image]

prasimix, 2021-02-01:
Fixed