eez-open/modular-psu-firmware #118: Log configuration GUI

fietser28, 2020-11-08:
The log configuation menu is missing the possibility to specify a trigger. 
The menu is also not consistent with the other menu's of the bottom row: 
- It is a popup/overlay menu up instead a full screen menu
- The Cancel button doesn't really cancel your changes it just goes back to the main menu. There is no conformation/cancel option like on other settings menu's.

Below a suggestion how this can be solved:
[image]

The changes I made to the screen:
- full screen with a container at the bottom with a home, OK and cancel option. Just like other screens
- removed cancel button, not needed anymore.
- added a trigger button. In my idea this should to go a screen where you set the trigger options for the logging. This screen should be almost the same as the trigger screen of a channel.
- a blank filename causes a filename with a timestamp. Show it like this.

If triggering of data logging is recurring maybe there should also be options how to handle the filename (if it is not a timestamp): replace the file, add an auto increment number or maybe append to the same file?
[image]

mvladic, 2020-11-30:
Proposed new SCPI commands and queries:

- `SENSe:DLOG:FUNCtion:DIGital[:INPut] {pin}, {enable}[, {channel}]`
- `SENSe:DLOG:FUNCtion:DIGital[:INPut]? {pin}[, {channel}]`

prasimix, 2021-01-29:
The last set color is used to mark all channels. For example, I set the magenta on the DCP module in slot 2 and the same color is used to display the DCP module parameters on slot 1:

[image]

prasimix, 2021-02-01:
Fixed

closed 2021-02-01 by prasimix