eez-open/modular-psu-firmware #123: Low current values presentation in DLOG viewer

prasimix, 2020-11-18:
Current values in DLOG viewer are rounded to 1 mA which is not acceptable for data acquired in low current range.

fietser28, 2020-11-21:
I think only the settings of the scales is limited. If I use view data option from the cap reform script, values below 1mA are shown:

[image]
In this case the script set the min/max values of the current from 0.0 to 200uA and that works. Not the offset and div are rounded to 0. If uA was shown it would probably say -200 (or -0.2 mA). Div is same.

If I start changing the scales (Div:) in the GUI it immediately limits me to the 1mA.

Tested on the unreleased 1.6 version.

prasimix, 2021-01-29:
Cap reform script returns error. It requires an investigation before testing the reported problem can continue: 

[image]

prasimix, 2021-01-29:
Test configuration: DCP in slot 1 and 2, SMX in slot 3. It is enough to tap on `Capacitor Max Voltage:` field to get previously reported error.
[image]

prasimix, 2021-01-29:
I didn't use the latest script version, but on v1.3 we still have some issue with "phantom" OVP trip

prasimix, 2021-02-02:
Fixed

closed 2021-02-02 by prasimix