eez-open/modular-psu-firmware #130: Diode tester script stop issue

prasimix, 2020-12-08:
Selecting DLOG Stop while Diode Tester script is running will be ignored or could cause restart.

prasimix, 2021-01-29:
Interrupting the script no longer causes a restart. The question remains whether we want to have a message like this in case of premature termination, or something that is more “graceful”, like "Script terminated"

[image]

mvladic, 2021-02-03:
> Interrupting the script ...

Actually, you didn't interrupt the script explicitly - it was side effect of stopping DLOG trace after which `DLOG:TRACE:DATA` failed and this failure wasn't handled inside the script hence "`Uncaught script exception!`"

> The question remains whether we want to have a message like this in case of premature termination, or something that is more “graceful”, like "Script terminated"

I think not, because "`Uncaught script exception!`" is exactly what happened here.

In this particular case we could catch `DLOG:TRACE:DATA` error and reported something like "DLOG stopped, breakdown voltage not found".

prasimix, 2021-02-03:
> In this particular case we could catch `DLOG:TRACE:DATA` error and reported something like "DLOG stopped, breakdown voltage not found".

OK

prasimix, 2021-02-03:
Fixed, affected scripts: CapReform (upgraded to 1.5), Curve Tracer (upgraded to 1.4) and Diode Tester (upgraded to 1.3).

closed 2021-02-03 by prasimix