eez-open/modular-psu-firmware #134: Static IP address not using netmask, gateway, DNS

MustBeArt, 2020-12-14:
I am seeing a problem with Ethernet networking on BB3 v1.5 when using a static IP address (DHCP disabled). NTP fails completely: the NTP test tries forever, but no packets come out of the Ethernet port, not even a DNS query to look up the NTP server name. Inbound connections and pings from the same /24 subnet as the BB3 work fine. Connections and pings from outside the /24 but inside the /16 go unanswered, even with the netmask set to 255.255.0.0. Connections from outside the /16 also go unanswered. It's as if the netmask, gateway, and DNS settings are being ignored (plus the NTP test function has a very long timeout, or none at all). With DHCP enabled, everything seems to work as expected (though I haven't tested very much with DHCP).

MustBeArt, 2020-12-17:
I built and tested master as of 2bb6ec7. DNS now works with a static IP address, and that solves the problem with NTP. Excellent!

There is still a peculiarity with ping that may merit investigation. I'm not sure the behavior is wrong, but it's different from what a Rigol instrument does. Here's the scenario: an ICMP ping request arrives from outside the LAN. The BB3 broadcasts an ARP request for the source IP address of the ping. If no ARP reply is received, the BB3 continues to rebroadcast the ARP request, and never sends a ping reply. Additional ping requests may arrive, but the BB3 never replies unless it receives an ARP reply.

The Rigol instrument also broadcasts an ARP request when it receives a ping request. However, when a second ping request arrives, it goes ahead and replies to the source MAC from the ping request. In fact it sends replies to both of the ping requests. This seems reasonable to me, in that it is making use of information that's already available instead of asking for it again via ARP. However, I'm not familiar enough with the RFCs to put my finger on language that permits or prohibits this technique.

In my case, I'm using a Wireguard VPN server to isolate the test equipment from the internet. I'm able to work around the BB3's current behavior by enabling proxy ARP on the VPN server. Proxy ARP seems like it makes sense in this case, but it doesn't seem to be required with the other instruments.

mvladic, 2021-01-29:
How can I test this?

fietser28, 2021-01-29:
@MustBeArt  describes a typical (default) route bug OR configuration error/missing default route or netmask setting.

BB3 should send an ARP request for the address of the default router (if it is not already in its cache) NOT for an address outside the local network. The BB3 determines wich MAC address to send the reply to based on it's own IP address and netmask. It should send responses to the MAC address directly (local network) or use the default router (gateway) MAC address. This is generic IPv4.

@mvladic If you don't have a network with seperate IP ranges you can create it with any machine using multiple network cards OR wait and I can try to reproduce the problem @MustBeArt describes.

mvladic, 2021-01-29:
I do have a network with separate IP ranges, still I could use some help how to reproduce the problem ;)

fietser28, 2021-01-29:
@mvladic You also need seperate VLAN's or cables otherwise you're test might be incorrect.

I just tested it with DHCP: Put the BB3 in a different VLAN and subnet. I'm able to ping and connect to the BB3 from a different subnet. I did not sniff al the packets, but it seems to behave properly.

I did not test static IP: Somehow the input fields don't allow me to input anything useful?

Still running you're 1.6-test firmware from november.

mvladic, 2021-01-29:
@fietser28 Uh, yes, that version had a broken keyboard, sorry :( Here is the latest firmware where keyboard works:

[bb3-v1.6.dfu.zip](https://github.com/eez-open/modular-psu-firmware/files/5894610/bb3-v1.6.dfu.zip)

At the beginning you will be asked which version of MCU you have. I assume you have R2B4.

fietser28, 2021-01-29:
Tested with this firmware and static IP/netmask/gateway: This works in my case as I would expect from a IPv4 device.
Used a port on a managed switch with port mirroring to see traffic from an to BB3:
- I tested both DHCP and static IP. Both in 10.* range with a /24 netmask (255.255.255.0). 
- Tested DNS server on local subnet and internet
- Tested TCP connections (port 5025) and ping from local subnet and different subnet (using VLANS)
All works fine (one I fixed firewall rules on my network...) and packets look normal to me.

Sorry, I can't reproduce the problem.

prasimix, 2021-02-03:
Fixed for now

closed 2021-02-03 by prasimix