eez-open/modular-psu-firmware #140: Micropython NameError

willg, 2021-01-15:
After running my script, [CC-CV Battery Charge.zip](https://github.com/eez-open/modular-psu-firmware/files/5819887/CC-CV.Battery.Charge.zip), I sometimes get `NameError`s for both my script and the official (unmodified) "Hello World" program.  The trace for the "Hello World" program is:
```
NameError: name not defined
    File"<stdin>", line 30 in main
    File "<stdin", line 39 in <module>
```

The behavior seems slightly different between release v1.5 and dev v1.6.  In v1.5 a reboot initiated through the on-screen gui would often fix the problem and then my script would work ok seemingly indefinitely until the next the BB3 was turned off.  In v1.6, "Hello World" will work ok, but then after running my script the micropython interpreter seems "poisoned", and no scripts will run with out a `NameError`.  

After a little research, it appears micropython doesn't implement the [del method](http://docs.micropython.org/en/latest/genrst/core_language.html#special-method-del-not-implemented-for-user-defined-classes) for user defined classes.  If the answer is "don't use classes" that is 100% ok.  The use of classes for this script is largely unnecessary, I was just trying to tidy up what otherwise would be global state.  

For reference, this was also briefly discussed in discord [here](https://discord.com/channels/511131497801973761/768242089388933140/799050730543251457).

mvladic, 2021-01-29:
I am getting slightly different result when I run your script:

```
2021-01-28 22:04:54 INFO Script started: CC-CV Battery Charge
2021-01-28 22:04:55 DEBUG Traceback (most recent call last):
2021-01-28 22:04:55 DEBUG   File "<stdin>", line 35, in <module>
2021-01-28 22:04:55 DEBUG   File "<stdin>", line 36, in charger
2021-01-28 22:04:55 DEBUG NameError: name not defined
2021-01-28 22:04:55 INFO Script ended: CC-CV Battery Charge
```

So, error is in the line 36:

```
...
35: class charger():
36:     total_seconds
37:     total_amp_hour
...
```

Variables `total_seconds` and `total_amp_hour` are class variables and must be assigned to some value. So, the following will pass:

```
...
35: class charger():
36:     total_seconds = 0
37:     total_amp_hour = 0
...
```

The same goes for these:

```
class calculator():
    cell_count = 0
    cell_charge_voltage = 0
    battery_capacity = 0
    c_rate_charge = 0
    c_rate_term = 0
    calc_chemistry = 0
```

But, looking at your code I think you don't need to declare these variables as class variables at the first place as you are using them as instance variables (through `self.`).

With these changes and also some changes in the firmware - we now reinitialize MP engine on every script run - I was able to successfully run your script and other scripts multiple times without errors.

mvladic, 2021-01-29:
Also, don't forget to call `SCIP:DLOG:CLEar` command at the beginning of DLOG settings definition to be sure that you are starting with the clean slate (for more information check the #116):

```
scpi("SENS:DLOG:CLE")
```

willg, 2021-01-29:
Ah, I see, that all makes sense.  

Re-initializing the MP engine is a good idea.  Sounds like during my development old instances of variables with the same name were left around in the interpreter and causing the behavior I saw.  

 Thanks for taking a look at this!

prasimix, 2021-01-31:
Fixed for now.

closed 2021-01-31 by prasimix

willg, 2021-02-02:
As further evidence @mvladic correctly identifed the issue and this is fixed, I updated my script to use properly initialized instance variables and tested on a build prior to [cf475ca](https://github.com/eez-open/modular-psu-firmware/commit/cf475cac7ffd5f713336de79ab8ab5805b047a6c).  I no longer get any `NameError` on my script or any others.  

Thanks again!