eez-open/modular-psu-firmware #144: Encoder increment/decrement leads to confusing display

duncairn, 2021-01-22:
When the encoder is used to set the voltage or current, the display removes trailing 0s and is then justified. This leads to the displayed setting moving about horizontally and the lower digits can be confusing. For instance, when set to increment by 25mV the display may go: 4.1V, 4.125V, 4.15V, 4.175V, 4.2V, etc. Since the display is center justified, the '4.1' part moves left and right and it's difficult to keep track of the mV values incrementing since, from step to step, the same column may be showing tens of mV or hundreds of mV. Even unchanging columns appear to change since they move left and right at each step.

Contrast with setting the step to 100mV, where the display goes 4.1V, 4.2V, 4.3V, etc. In this case it is easy to see which column is changing and how close to the desired setting it is getting.

There are two possible fixes:

    When changing the display, use left justification.

    When changing the display, don't remove trailing digits (that is, have it display, for instance, 4.100V).

After setting, either with a short timeout or because the setting has been accepted, the display can revert to center justification and trailing 0s removed.

(Sorry for posting in the wrong place originally!)

electrokean, 2021-01-23:
Maybe the rule for removing trailing zeros should depend on the encoder step size.
So for example with voltage setting, if the encoder step is 5mV or 25mV, then you would always show 3 decimals.
If the encoder step is 100mV or greater, then just show 1 decimal unless there are more significant digits available.
Might need some extra through for the Auto step size.

prasimix, 2021-01-23:
Thanks for your input. Clearly, something will need to be done here. A lot of people have already complained and the first step has been made in the new GUI for MIO168 in a way that the number of digits is fixed to 5, as you can see on screenshot below: ch1 is in 240.00 V range, ch2 in 48.000 V range, ch3 in 1.2000 A and ch4 in 10.000 A range.
Perhaps it would be best to do the same with other modules: to fix the number depending on the accuracy of the module.

[image]

duncairn, 2021-01-23:
> Maybe the rule for removing trailing zeros should depend on the encoder step size.

That would seem to be good, yes.

Looking at the image @prasimix has included, AIN1 and AIN3 have different number of digits but are in the same horizontal position. The issue described here thus wouldn't affect those, and the obvious difference in accuracy is nice Ux cue to boot.

prasimix, 2021-02-02:
See also #88