eez-open/modular-psu-firmware #145: With Rprog in DCP405 Power limit should be inactive in display and OPP setting

fietser28, 2021-01-24:
When using the Rprog interface of the DCP405 the power limit is still shown as active in the GUI, but this is not enforced:
[image]
In this screenshot on channel 2 36W is drawn, where the UI gives the impression 5W is the limit. I think this should be graded out, 0W or not visible in this case (just like the set voltage).

OPP still works correctly, but setting the OPP trip limit complains about the (not present) power limit too:

[image]

prasimix, 2021-02-02:
* P_LIM option should be changed to read-only and change to `RP`
* Same as above, change U_SET value that is already read-only for 0.0 V (that is somewhat confusing) to `RP`

prasimix, 2021-02-02:
Fixed

closed 2021-02-02 by prasimix