eez-open/modular-psu-firmware #146: Change Protection ("shield") icon to channel settings icon on power modules

prasimix, 2021-01-26:
- [ ] Change Protection ("shield") icon to module/channel settings icon on power modules to be in line with other modules look
- [ ] Change Setings to shield icon only in case of protection trip
- [ ] Change option "Clear and Disable" on protection trip page to "Settings" which will display Protection settings

prasimix, 2021-01-29:
Fixed

fietser28, 2021-02-11:
1.6-RC has different implementation: last bullet is implemented different? 
A trip shows a dialog with only a 'close' button, doesn't present the choice go to the relevant menu ("settings")?

[image]

prasimix, 2021-02-12:
Fixed

closed 2021-02-12 by prasimix