eez-open/modular-psu-firmware #148: DCM220 doesn't round off low power values

fietser28, 2021-01-31:
The DCM220 current measurements <250mA are not shown on DCM220. Power is stil shown for low currents. It should also be shown as <... W (depending on measured voltage * 0,25 A )?

On attached screenshot (CH3 & CH4 are DCM220) CH3 is without anything connected. CH4 has a small load connected.

[image]

It think it should show:  
CH3: <0,615W    (2,46V   * 0,25A)
CH4: <5W           (20,05V * 0,25A)

prasimix, 2021-01-31:
The same has to be done for DCM224, too.

prasimix, 2021-02-01:
Fixed

closed 2021-02-01 by prasimix

fietser28, 2021-02-11:
This created a new issue #158 (the roundoff itself is working).