eez-open/modular-psu-firmware #149: Clear protection not always shown in red

fietser28, 2021-01-31:
I like the new clear protection look, but for me there are two inconsistencies (for some **** users like me) in the GUI:

1. I perceive the new way as 'follow the red items' (red shield, red menu item). The final 'clear protection' button should be red also to make this flow intuitive (to me). (I missed it the first time).
2. If I go into the menu's and DON'T clear the protection but for example turn the protection off and save it. Then the protection is still on, the red shield is shown on the main screen, but the corresponding menu item of the protection is not red anymore.  

Below are screenshots of my journey:
[image]
I tripped OPP, changed icon and red are very clear, nice. I know were to go: push it.

[image]
Very clear where the problem is, red menu item...

[image]
Here I missed the red item (clear protection). = item 1 above.

[image]
Know it gets interesting: If I turn of OPP protection (not clearing the protection/trip)...

[image]
Back to the main menu: OPP trip is still there.

[image]
Into the menu: The red menu is not there anymore! = item 2 above.
The OPP trip is still active.

I did not test this flow for the other protections (voltage, current, temp).
Running 1.6 development firmware.

fietser28, 2021-01-31:
Found 2 more similar inconsistencies in different view:

[image]
Screen looks good, but pressing OPP in this scenario (it is tripped/RED) will bring me to the general protection menu. If OPP is not tripped, pressing OPP will jump directly to the Power protection menu (= desired action).

[image]
Performing the same flow as above: Going to the power protection menu, not clearing the trip but disabling the OPP. and returning to the main screen: the trip is still present and indicated by the shield at the top, but the OPP is not RED anymore.

fietser28, 2021-01-31:
Small addition to item 2: Setting the power limit ON again (still without clearing the OPP trip) will make the menu item 'power limit and OPP' red again when you visit it. 
changing other values like the level of the power limit doesn't cause this. 
It appears the menu only turns red if there is a power trip AND the power limit is on. This appears false because you can disable the protection without clearing the trip.

prasimix, 2021-02-01:
I think we need to clear the trip of the protection we want to turn off. This will avoid the need to clear the trip afterwards.

prasimix, 2021-02-01:
I managed to enter current calibration with OCP eanbled, and got OCP trip:

[image]

prasimix, 2021-02-01:
Fixed for now

closed 2021-02-01 by prasimix