eez-open/modular-psu-firmware #150: OVP changes in DProg mode

prasimix, 2021-02-03:
- [ ] If OVP HW is active when entering RProg mode, it should be changed to OVP SW because HW will immediately trip due to the principle it works on. When exiting RProg mode, it can return to OVP HW. 
- [ ] Reduce OVP trip level from 40.5 to 40.05 and prohibit setting more than 40.05 V.

prasimix, 2021-02-04:
Fixed

closed 2021-02-04 by prasimix

fietser28, 2021-02-11:
If OVP is OFF (OVP active = OFF) and you enable RProg, OVP is turned ON (software). 
(If I turn RProg OFF then OVP stays ON)
Is this intentional?

When Rprog is tuned on it should:
- change OVP type from HW to SW only if OVP is active HW
- change trip level to 40.05 only if OVP is ON (and being changed to SW). 

Tested on 1.6-RC by going thru menu's. I assume the OVP menu display the settings correctly.

prasimix, 2021-02-12:
Idea here is to force SW OVP as protection against too high RProg voltage that could lead to module damage if applied for prolonged period of time. Question is if we want to set OVP in previous state or not when RProg is turned off.