eez-open/modular-psu-firmware #160: Step settings in channel trigger display not updated from changes ramp & delay overview

fietser28, 2021-02-11:
If I go into the trigger menu, select Step and set value (note the 5.2V):

[image]

Then press 'ramp & delay overview' and in this display change values (e.d. voltage to 5.5):

[image]

IF I then press the OK button, going back to the original trigger menu, the old values are displayed: 

[image]

Leaving the trigger menu and returning to it shows the updated value (5.5V). So I think the display is just not refreshed.

Tested on 1.6-RC

prasimix, 2021-02-12:
I cannot reproduce this issue, perhaps because I'm testing it on 1.6-RC2

fietser28, 2021-02-15:
Just tried it with 1.6-RC2. Problem is still there. How can I explain it more clearly? Upload a movie in Discord?

Basically the values are not updated on the screen 'chx: trigger settings' if you close the 'ramp & delay overview' screen.

prasimix, 2021-02-15:
It should be fixed now in 14cdfee

fietser28, 2021-02-15:
Tested with RC3. It is fixed. Thanks.

prasimix, 2021-02-15:
Fixed

closed 2021-02-15 by prasimix