eez-open/modular-psu-firmware #171: Clear DLOG REMark and X axis label when DLOG is started through GUI

mvladic, 2021-03-01:


prasimix, 2021-06-10:
Test procedure needed.

mvladic, 2021-06-10:
First, make some DLOG using SCPI, for example:

```
SENS:DLOG:TRAC:X:UNIT VOLT
SENS:DLOG:TRAC:X:SCAL LIN
SENS:DLOG:TRAC:X:STEP 1
SENS:DLOG:TRAC:X:RANG:MIN 0
SENS:DLOG:TRAC:X:RANG:MAX 40
SENS:DLOG:TRAC:X:LAB "X-Axis label"

SENS:DLOG:TRAC:Y1:UNIT AMPE
SENS:DLOG:TRAC:Y:SCAL LIN
SENS:DLOG:TRAC:Y1:RANG:MIN 0
SENS:DLOG:TRAC:Y1:RANG:MAX 5
SENS:DLOG:TRAC:Y1:LAB "Y-Axis label"

SENS:DLOG:TRAC:REM "Testing #171"
INIT:DLOG:TRAC "/Recordings/test_171.dlog"
SENSe:DLOG:TRAC 1
SENSe:DLOG:TRAC 2
SENSe:DLOG:TRAC 3
SENSe:DLOG:TRAC 4
SENSe:DLOG:TRAC 5

ABOR:DLOG

MMEM:UPLOAD? "/Recordings/test_171.dlog"
```
Here is the result:

[image]

Now, make some DLOG using GUI and transfer it to the Studio:

[image]

No DLOG remark or Axis labels should be visible from previous DLOG.

prasimix, 2021-06-10:
The test passed, however I found on the screen an abundance of decimals for DLOG generated using SCPI commands, which need to be fixed:

[image]

prasimix, 2021-06-23:
Fixed

closed 2021-06-23 by prasimix