eez-open/modular-psu-firmware #174: Step menu allows negative delay setting with encoder

fietser28, 2021-03-30:
If you use the encoder to change the delay in the step menu of the DCP405 it doesn't honour the minimum value (4ms?), but allows you to even to negative. See screenshots.

[image]
[image]

Entering the value with virtual keyboard enforces the correct lower limits.

prasimix, 2021-03-30:
I can reproduce even worse behavior: it will restart my BB3 (tested with MCU r3B3, latest unpublished 1.7 firmware)

prasimix, 2021-06-10:
Fixed

closed 2021-06-10 by prasimix