eez-open/modular-psu-firmware #176: DLOG view doesn't remember settings

fietser28, 2021-03-30:
If I adjust the log viewer to zoom in on something specific (setting offsets and divs to what I want to see), then press the home icon (going out of dlog viewer) then go back in log viewer, alle settings are gone. :-(

Example:
1. Setting up my nice view:
[image]

2. Press the home button, going to the main BB3 screen
3. Press the dlog viewer button again, gives me:
[image]

There are 2 different situations when returning to the log viewer:
- In this case (going back to same data) I think the settings should be retained. 
- If viewing new data, maybe a question should be asked to auto scale or retain previous settings. This is useful in case you perform a new measurement (using triggers) and want to look at the same point in time and scales.

mvladic, 2021-06-11:
This is the rule: If the new dlog file has the same X-Axis and Y-Axes parameters as the previous dlog file then settings will be retained.

Same X-Axis and Y-Axes parameters means the following:

- X-Axis range is the same
- X-Axis scale type is the same
- X-Axis unit is the same
- X-Axis step is no different than 5%
- has the same number of Y-Axis and each Y-Axis has the same: channel index, range and unit

fietser28, 2021-06-11:
Sounds good. Have to get a feeling for it in practical use, but that will be a different issue then.

prasimix, 2021-06-12:
Fixed

closed 2021-06-12 by prasimix