eez-open/modular-psu-firmware #198: Programing LIST refactoring

prasimix, 2021-05-11:
Programming LIST currently support only power modules (DCP, DCM). We have to add support for SMX46 AOUT channels and MIO168 DOUT and AOUT channels. LIST editor has to be modified to work with multiple channels (on different modules).

mvladic, 2021-05-27:
- Trigger mode (new option: FUNCgen)
    - For power channels
		- `SOURce:VOLTage:MODE {FIXed|LIST|STEP|FUNCgen}`
		- `SOURce:VOLTage:MODE?`
		- `SOURce:CURRent:MODE {FIXed|LIST|STEP|FUNCgen}`
		- `SOURce:CURRent:MODE?`

    - For MIO and SMX AOUT
		- `SOURce:MODE FIXed|FUNCgen`
		- `SOURce:MODE?`

    - For MIO DOUT
		- `SOURce:DIGital[:OUTPut]:MODE {pin}, {FIXed|FUNCgen}`
		- `SOURce:DIGital[:OUTPut]:MODE? {pin}`

- Function generator waveform shape
    - For power channels
		- `SOURce:VOLTage:FUNCtion:SHAPe {DC|SINe|TRIangle|SQUare|PULSe|SAWTooth}`
		- `SOURce:VOLTage:FUNCtion:SHAPe?`
		- `SOURce:CURRent:FUNCtion:SHAPe {DC|SINe|TRIangle|SQUare|PULSe|SAWTooth}`
		- `SOURce:CURRent:FUNCtion:SHAPe?`

    - For MIO and SMX AOUT
		- `SOURce:FUNCtion:SHAPe {DC|SINe|TRIangle|SQUare|PULSe|SAWTooth}`
		- `SOURce:FUNCtion:SHAPe?`

    - For MIO DOUT
		- `SOURce:DIGital[:OUTPut]:FUNCtion:SHAPe {SQUare|PULSe}`
		- `SOURce:DIGital[:OUTPut]:FUNCtion:SHAPe?`

- Function generator waveform other parameters
    - For power channels
		- `SOURce:VOLTage:SWEeap::FREQuency {frequency}`
		- `SOURce:VOLTage:SWEeap::FREQuency?`
		- `SOURce:VOLTage:SWEeap::AMPLitude {amplitude}`
		- `SOURce:VOLTage:SWEeap::AMPLitude?`
		- `SOURce:VOLTage:SWEeap::OFFSet {offset}`
		- `SOURce:VOLTage:SWEeap::OFFSet?`
		- `SOURce:VOLTage:SWEeap::PULSewidth {pulse_width}`
		- `SOURce:VOLTage:SWEeap::PULSewidth?`
		- `SOURce:CURRent:SWEeap::FREQuency {frequency}`
		- `SOURce:CURRent:SWEeap::FREQuency?`
		- `SOURce:CURRent:SWEeap::AMPLitude {amplitude}`
		- `SOURce:CURRent:SWEeap::AMPLitude?`
		- `SOURce:CURRent:SWEeap::OFFSet {offset}`
		- `SOURce:CURRent:SWEeap::OFFSet?`
		- `SOURce:CURRent:SWEeap::PULSewidth {pulse_width}`
		- `SOURce:CURRent:SWEeap::PULSewidth?`

    - For MIO and SMX AOUT
		- `SOURce:SWEeap::FREQuency {frequency}`
		- `SOURce:SWEeap::FREQuency?`
		- `SOURce:SWEeap::AMPLitude {amplitude}`
		- `SOURce:SWEeap::AMPLitude?`
		- `SOURce:SWEeap::OFFSet {offset}`
		- `SOURce:SWEeap::OFFSet?`
		- `SOURce:SWEeap::PULSewidth {pulse_width}`
		- `SOURce:SWEeap::PULSewidth?`

    - For MIO DOUT
		- `SOURce:DIGital[:OUTPut]:SWEeap::FREQuency {frequency}`
		- `SOURce:DIGital[:OUTPut]:SWEeap::FREQuency?`
		- `SOURce:DIGital[:OUTPut]:SWEeap::PULSewidth {pulse_width}`
		- `SOURce:DIGital[:OUTPut]:SWEeap::PULSewidth?`

- Selection of voltage or current
    - For MIO AOUT1 and AOUT2
		- `SOURce:FUNCtion:MODE {VOLTage|CURRent}`
		- `SOURce:FUNCtion:MODE?`

mvladic, 2021-05-28:
- Example for power channel

```
INST (@101)

VOLT:MODE FUNC
VOLT:FUNC:SHAPe SIN
VOLT:SWE:FREQ 50
VOLT:SWE:PHAS 90
VOLT:SWE:AMPL 10
VOLT:SWE:OFFS 20

CURR:MODE FUNC
CURR:FUNC:SHAPe PULS
CURR:SWE:FREQ 10
CURR:SWE:PHAS 30
CURR:SWE:AMPL 1
CURR:SWE:OFFS 2
CURR:SWE:PULS 75

INIT:IMM
```

- Example for MIO AOUT

```
INST (@307)

SOUR:MODE FUNC
SOUR:FUNC:SHAPe TRI
SOUR:FUNC:MODE CURR
SOUR:SWE:FREQ 250
SOUR:SWE:PHAS 0
SOUR:SWE:AMPL 0.005
SOUR:SWE:OFFS 0.01

INIT:IMM
```

- Example for SMX AOUT

```
INST (@201)

SOUR:MODE FUNC
SOUR:FUNC:SHAPe SQU
SOUR:SWE:FREQ 250
SOUR:SWE:PHAS 60
SOUR:SWE:OFFS 5
SOUR:SWE:AMPL 3

INIT:IMM
```

- Example for MIO DOUT

```
INST (@302)

SOUR:DIG:MODE 1, FUNC
SOUR:DIG:FUNC:SHAP 1, PULS
SOUR:DIG:SWE:FREQ 1, 100
SOUR:DIG:SWE:PHAS 1, 0
SOUR:DIG:SWE:PULS 1, 25

INIT:IMM
```

fietser28, 2021-05-28:
Please consider adding the complementary functions:
- period (= 1/frequency) 
- min/max (equivalent of amplitude/offset)
- delay (equivalent of phase at a given frequency)
- dutycycle (equivalent of pulse width at a given frequency)

period, delay and duty cycle  are nice to haves for SCPI commands (in a script/program you can do the calculation yourself), but I think in the user interface it is more important to be able to switch between them.

min/max is important if you want to change and existing output: You want to be able to sweep the max value of a sine or pulse by setting 1 value instead of 2 (risking a peak value in between the 2 commands).

Similair if I would use the encoder to change a value live: min/max is very different from offset/ampl: Changing max value allows you to slowly/easily increase/decreate a pulse from e.g. 0-3.3V to 0-5V by rotating the know. With offset/ampl you constantly have to change 2 values not 1.

As an example: My Siglent AWG allows to me to choose in the UI between displaying and setting:
- frequency or period
- amplitude/offset or min/max
- pulse width or dutycyle (if applicable with the waveform)
I really like that. In general when working with digital level my mind is in 0-X volt-mode (min/max) and with analog signals it is more amplitude/offset. But that is just me.

For the user interface this is analog to what i described in #175, but that is just for displaying this also is for setting.

mvladic, 2021-05-28:
Could you explain how it works on Siglent AWG (maybe post some screenshots)? Is it global setting or you can select for each channel different setting?

fietser28, 2021-05-28:
Here are some screenshots. Below the screen are a row of 6 buttons. There function corresponds to the function on the bottom of the display. Blue is the active setting.

As an example I use a pulse.

1. Note the frequency button (blue) and the frequency is also displayed. Entering a number (or turning the encoder) is entering/changing the frequency, Hz, kHz, MHz,..
[image]

2. Now I pressed the frequency button twice. The AWG shifts to 'period mode'. It doesn't show the frequency but the corresponding period (1/f) in the same place where frequency was before. If I would enter a number now, it would be a period in s/ms/us/ns.
[image]

3. Now looking at the amplitude/offset vs. highlevel/lowlevel. In screenshot 1 and 2 the levels were shown as offset and amplitude. Pressing amplitude twice switches from amplitude/offset mode to high/low level mode. Note the change in button color and the replacement of amplitude and offset in highlevel and low-level in the list:
[image]

4. Same trick for pulsewidth vs. dutycyle:
[image]

In this example it is 100% UI, it doesn't change anything in the internal state of the AWG. If I change the period to 1s and switch to frequency it shows 1Hz (which corresponds to 1s). This goes for all the values. 

If I am in 'period mode' it is for both channels, it is a display view state not a channel state.

In case of the pulse it is important to notice: If I am in pulse width mode and I change the frequency, the pulse width stays the same (in s), but the duty cycle changes. If I am in duty cycle mode and I change the frequency the duty cycle stays the same but the pulse with changes. Completely logical, but worth noticing. Similair things happen if the rise an fall times are very large and I change the frequency to a value it would fit anymore, the rise an fall times are truncated to there possible limits.

The AWG doesn't have a great SCPI command set (in my opinion) so that is not a good reference. But I really like these different modes for the user. That is why I called this complementary. 

The only situation where I really need the complhigh and low levels is when switching/slowly moving from low-level 0 and high-level 5V to low-level 0V and high-level 3.3V. With only offset/amplitude this would require changing amplitude from 5.0 to 3.3 AND changing offset from 2.5 to 1.65. Doing this slowly/manually with an encoder is impossible with amplitude/offset mode, but works fine with high/low level mode by only changing highlevel.

I hope this clarifies things a bit.

prasimix, 2021-06-25:
Fixed

closed 2021-06-25 by prasimix