eez-open/modular-psu-firmware #215: Firmware updates from SD

fietser28, 2021-06-17:
I would like to be able to update the firmware of the BB3 from a file on the SD card. This issue is jus about the plain update method itself (probably an additional small step between the current bootloader (supporting dfu-util) and the main software).

I think this is an important enabler for other UI/user functions that download the firmware file to the BB3:
- directly via ethernet from GitHub
- via studio (USB or ethernet)

prasimix, 2021-06-17:
I believe that the first step here is to find out how to make custom bootloader and does it can be part of the firmware or factory default bootloader needs to be replaced.

goran-mahovlic, 2021-06-17:
As I know we cannot remove current factory default boot loader - so it stays. To have SD card bootloader we would need small application that reads SD card and writes data to memory address of STM32. 

Factory default bootloader jumps to custom bootloader address that will be at 0x08000000
custom bootloader checks if file is available - or button is pressed
custom bootloader loads app
custom bootloader jumps to app address that has offset (depending of bootloader size)

We can also find already written bootloader and check how it works...

https://github.com/akospasztor/stm32-bootloader/tree/master/projects/STM32L496-Discovery

goran-mahovlic, 2021-06-17:
This one looks promissing - https://github.com/feaser/openblt

prasimix, 2021-07-07:
Fixed partially, i.e. it is possible to detect and download new firmware indirectly from GitHub via EEZ Studio.

closed 2021-07-08 by prasimix