eez-open/modular-psu-firmware #228: Limits on function generator values are not properly enforced/set.

fietser28, 2021-06-26:
On FW 1.7-RC1:
When entering the amplitude or offset in the function generator it is possible to input impossible values/combinations.

The Min. / Max. ranges are properly checked and generate a warning on the DCP405 module. 
On the DCM220 this also creates impossible values (outside the range of the module).

In general each channel has an absolute min (Cmin) - max (Cmax) range.
When you enter:
-  the offset: The channel capabilities AND the current amplitude should both limit the range you can enter.
- the amplitude: The channel capabilities/limits AND the current offset should both limit the range you can enter.

Comparable rules apply to min/max.  

I did not check this with SCPI commands. I also didn't turn the channels on with these values.

[image]
[image]
[image]

prasimix, 2021-06-26:
"Impossible values" will results with error on start indicating that voltage, current or power limits have been exceeded.

fietser28, 2021-06-28:
Fixed.

closed 2021-06-28 by fietser28