eez-open/modular-psu-firmware #240: Autostart.py script support

prasimix, 2021-07-05:
Add possibility to run autostart.py MicroPython script on power up/restart if its exists in /scripts folder. Add the following:
- [ ] `DISPlay[:WINDow]:HOMe` command to display home page when script execution is done
- [ ] Option to enable/disable execution of the script, default is disabled
- [ ] Option to override script execution to stop execution of faulty script (e.g. if encoder switch is active during power up)

fietser28, 2021-07-05:
Combine with #214?

fietser28, 2021-07-05:
The first item also relates to #126. If the script doesn't start a dialog, but runs in background, the user cannot get control back?

To manage all these possiblities maybe an tile/menu to the systems setting where the user can manage autostart but can also manage running/closing scripts instead of  the overlay dialog?

mvladic, 2021-07-06:
SCPI commands:

```
- SCRipt:RUN {script_file_path}
    
  Runs script file.

  {script_file_path} is quoted string. Contains a full path to the script file, for example: "/Scripts/Hello world.py"

- SCRipt:RECall {script_file_path}
- SCRipt:RECall?

  Sets/selects auto run script. 

  {script_file_path} is quoted string. Contains a full path to the script file, for example: "/Scripts/AutoStart.py"

  Query returns selected auto run script or "" if none selected.

- SCRipt:RECall:CONFirmation
- SCRipt:RECall:CONFirmation?

  Enables/disables confirmation popup.

```

mvladic, 2021-07-06:
To stop currently running script that got stuck you can also press both encoder and user switch from the front panel.

fietser28, 2021-07-06:
Are SCRipt:RECall and CONFirmation part of the profile. E.g. selecting a different profile also sets a different script?

mvladic, 2021-07-06:
Yes.

fietser28, 2021-07-06:
Setting RECall and CONFirmation will be possible in the profile settings dialog? 
Is this a good place to add this there?

mvladic, 2021-07-06:
We added a new Script section in Settings:

[image]

[image]

mvladic, 2021-07-06:
This is in case when auto recall is not enabled:

[image]

prasimix, 2021-07-07:
Fixed

closed 2021-07-07 by prasimix

fietser28, 2021-07-07:
Small UI inconsistency: If you change the parameters in the dialog then there doesn't appear a 'cancel' / 'ok' button on the status bar to confirm. Most other settings menu's have this workflow. 

Tested on FW 1.7.1-RC1.

prasimix, 2021-07-07:
You're right. We are not 100% consistent on this issue. Perhaps we should open a new issue and list there all places where we're missing confirmation.