eez-open/modular-psu-firmware #258: Log Settings - Duration field ignores "min"

fusedFET, 2021-09-14:
On the log settings page, when selecting Duration and entering for example 2 minutes by clicking "2" and then "min", it sets the value to 1s.

Tested on 2 BB3 with firmware 1.7.1

[image]

fietser28, 2021-09-14:
"min" means minimal value (1s), not minute in the keyboard. By pressing s/ms you select the unit of the number. 

Currently you can't give minutes as a unit only seconds and milliseconds. Is that what you want to do enter minutes?

fusedFET, 2021-09-14:
Exactly,
I somehow expected "minutes" not "minimal value". Anyway, this resolves the issue. But still, I have at least 3 other people who thought it was "minutes". It seems to be a quite recurrent misunderstanding.

prasimix, 2021-09-21:
What if we put here `Min` instead of `min` ?

fusedFET, 2021-09-23:
It still feels a bit confusing.
I have seen _mim_ and _Mnm_ used sometimes instead of _Min_ or _min_.