eez-open/modular-psu-firmware #290: (MIO168) feature request: display analog input/output values in ‘sensor’ mode

fietser28, 2021-11-06:
Background:
When working with analog signals, these signals often correspond to a measurement (e.g. sensor) or other physical setting (e.g. speed, angle, temperature, ...). Most of the time the relation between the read/set voltage/current and the physical measurement is lineair (y=a.x + b in it simples form).

Request:
I would like the analog I/O channels to have a 'sensor mode' performing a (lineair) calculation for me and display it next of/instead of the actual voltage/current measurement/setting. 

This requires several things (per channel):
- display additional value and/or replace the voltage/current value and add a visual indicator.
- add ways to enter the translation factors and units: multiplication, offset and unit in UI and SCPI.

If this is in place, it will be easy to create a script/applet that set the multiplication/offset based on input from user and readings.

This is a big feature, take time to think it over and prioritise it. With the micro python and applet capabilities of the BB3, there are other ways to achieve this, but requires scripting for simple translation.

Some DMM's have this feature in a 'sensor mode', sometimes limited to temperature sensors.

prasimix, 2021-11-06:
Thanks, any reference to existing solution/instrument would be helpful in further planing of this feature.

fietser28, 2021-11-07:
On latest Siglent DDM's it looks like this:

Setup is via a CSV file or menu's:
[image]
- Type is the physical measurement type the DDM should perform (VDC, ADC, IDC, ADC,...) in our case this is fixed and based on the channel (@305)
- Name is a sensor name
- Unit: things like C/F/% or a letter 

Entering values is a table:
[image]

You can also name the sensor (Imon in this case and choose the unit of the sensor). In use it looks like this:
[image]
It also works in the trends and histogram readings:
[image]
It would be nice if you could also select the sensor in DLOG... :-)

NOTE: I'm measuring a voltage here being a sensor for a current.