eez-open/modular-psu-firmware #297: Strange output values after standby

fusedFET, 2021-12-02:
Hi,
I have found a bug related to standby.
Suppose the device is operating, and you go into standby.
After waking up, the system has reset the channel configurations. If you set it again, the voltage outputted physically is wrong and not the one shown on the BB3. Also the BB3 shows current is flowing, but there is no load attached.
Firmware: 1.7.3, 
HW: DCP405 R2B11

Here a small video waking up from standby:
The measured output voltage is 3.4878v using a multimeter, the two leads are not connected to anything. No current is flowing, but the BB3 says 5.5mA.
[Video](https://1drv.ms/v/s!AvQwX7vex5Mhpr9s-1McPDiXL0IhOQ?e=ZZyghp)

prasimix, 2021-12-02:
Please check if DP (down-programmer) is enabled on that channel.

fusedFET, 2021-12-02:
Yes, DP is enabled on this channel.

prasimix, 2021-12-02:
Hm, then probably something is wrong with DP. The easiest way to check it is to run a simple list with two values and dwell times of e.g. 10 ms or more and check how the output looks like when YT view is selected.

fusedFET, 2021-12-06:
Hello, sorry for the delayed response.
So here is the test with the LIST mode (3v, 5v, with 10 ms Dwell Time).
[image]

And physically, the output is:
[image]

With high at 4.939V and low at 2.86V. The timing is accurate. (10ms up, 10ms down).
There is no notable load besides a logic analyzer connected.
The measure has been done after the standby (which did again reset all settings).