eez-open/modular-psu-firmware #298: DCP405: triggers unwanted Standby

fusedFET, 2021-12-06:
When disabling a channel, the BB3 sometimes goes into standby.
Firmware: 1.7.3
HW: DCP405 R2B11

There is a large part of random to reproduce it, but I now have encountered it multiple times.

[Video](https://1drv.ms/v/s!AvQwX7vex5MhpsEIU7Si8z3VcWPu3w?e=f1olbi)

fusedFET, 2021-12-06:
Here is the event log when having the issue:
[image]

prasimix, 2021-12-06:
Ok, so you probably have this option enabled: 
[image]

fusedFET, 2021-12-06:
Correct, I have this setting enabled.

prasimix, 2021-12-06:
That's fine, that is the trip which bring them into standby mode due to that option activated. So it seems that some overshoot is present on the output enabled. Is it possible that you have load connected with longer cables on that channel?

fusedFET, 2021-12-07:
I do have some longer cables, about 40 cm.
Maybe in order to clarify the issue, it would be practical to have a message on screen when going into standby for this reason:
"OVP Triggered Standby..." ?

prasimix, 2021-12-07:
> I do have some longer cables, about 40 cm. Maybe in order to clarify the issue, it would be practical to have a message on screen when going into standby for this reason: "OVP Triggered Standby..." ?

Yes, that makes sense, otherwise it can look confusing (and scary) if the user forgets that he chose the option to go to standby because of the trip.