eez-open/modular-psu-firmware #305: MIO168+AFE3: AIN3+AIN4 Power calculation is off?

fietser28, 2022-06-05:
[image]
P2 should read 0,4889 W.
DC power source is used, all numbers are stable.

AC analysis is correct:
[image]

fietser28, 2022-06-05:
Power function is using uncalibrated values.

MartinNPN, 2025-02-19:
Couldn't this closed after https://github.com/eez-open/modular-psu-firmware/pull/306 is merged ?

fietser28, 2025-02-19:
> Couldn't this closed after [#306](https://github.com/eez-open/modular-psu-firmware/pull/306) is merged ?

Yes, it is in the master branch. But I don't think it is in the current release (1.7.x). Not sure though.